robofit/arcor2_areditor #25: NewObjectType RPC error ( "problem": null)

ZdenekM, 2020-03-24:
```
Sending data to server: {
  "id": 24,
  "request": "NewObjectType",
  "args": {
    "abstract": false,
    "base": "Generic",
    "built_in": false,
    "description": "",
    "disabled": false,
    "needs_services": [],
    "object_model": {
      "type": "Box",
      "box": {
        "id": "BoxZZ",
        "size_x": 0.1,
        "size_y": 0.1,
        "size_z": 2.0
      }
    },
    "problem": null,
    "type": "BoxZZ"
  }
}
UnityEngine.Debug:Log(Object)
Base.WebsocketManager:SendDataToServer(String, Int32, Boolean) (at Assets/BASE/Scripts/WebsocketManager.cs:152)
Base.<CreateNewObjectType>d__47:MoveNext() (at Assets/BASE/Scripts/WebsocketManager.cs:461)
System.Runtime.CompilerServices.AsyncTaskMethodBuilder:Start(<CreateNewObjectType>d__47&)
Base.WebsocketManager:CreateNewObjectType(ObjectTypeMeta)
Base.<CreateNewObjectType>d__124:MoveNext() (at Assets/BASE/Scripts/GameManager.cs:493)
System.Runtime.CompilerServices.AsyncTaskMethodBuilder`1:Start(<CreateNewObjectType>d__124&)
Base.GameManager:CreateNewObjectType(ObjectTypeMeta)
<CreateNewObjectType>d__22:MoveNext() (at Assets/2D_EDITOR/Scripts/NewObjectTypeMenu.cs:126)
System.Runtime.CompilerServices.AsyncVoidMethodBuilder:Start(<CreateNewObjectType>d__22&)
NewObjectTypeMenu:CreateNewObjectType()
UnityEngine.EventSystems.EventSystem:Update() (at /home/zdenal/Unity/Hub/Editor/2019.3.0f6/Editor/Data/Resources/PackageManager/BuiltInPackages/com.unity.ugui/Runtime/EventSystem/EventSystem.cs:377)

```

Kapim, 2020-03-24:
Should be fixed by commit [https://github.com/robofit/arcor2_editor/commit/19c5e87df8b0a5edcab0d3883485e5ce3026b8bf](19c5e87df8b0a5edcab0d3883485e5ce3026b8bf)

Kapim, 2020-03-24:
Fixed by #30

closed 2020-03-24 by Kapim