robofit/arcor2_areditor #36: Right click on AP inside collision object (safty collision box)

Kapim, 2020-03-30:


Kapim, 2020-03-30:
@NelaMm What do you mean by this issue? Is it bug or enhancement request?

NelaMm, 2020-03-30:
It is enhancement request. For example: We add collision box and we have action point inside collision box (grab something from box), AP is visibile when box visiblility is on 50% but on right-click I edit box not AP inside. It will be good when we can edit it (for example on some else type of click or another way)

Kapim, 2020-03-31:
Ok, now i see. At the moment, you can globally turn off all interaction with action objects using Action objects settings menu (the gear button on bottom panel)
[image]

In this menu you can set if action objects should be visible and interactive. Below this settings you can find list of all action objects in scene/project, which can be used to open menu of all objects, regardles of their visibility or interactivity.
[image]

Do you think that it would be usefull to enable/disable interactivity of each object individually in action object menu or is this global approach sufficent?

Another way how to interact with AP inside object is using menu of action object, which contain list of its action points: 
[image]

NelaMm, 2020-03-31:
Sorry I didn't noticed this option.  I think a global approach is sufficient for now.

Kapim, 2020-04-01:
Ok, no problem, these features will be covered in user manual in the future.

closed 2020-04-01 by Kapim