robofit/arcor2_areditor #45: Generate user freindly action id/names

NelaMm, 2020-04-03:
Generate user freindly action names for example composed from AP action parameters - for better orientation for user and enable modify that name

Kapim, 2020-04-15:
We'll try to find something out

Kapim, 2020-05-12:
We are just thinking of this and we don't think that using parameters to generate name is not a good idea, because:

1. At the moment when the name should be generated, the actual values of parameters are not known
2. Parameters are subjects of change and you probably don't want to change name of action in case of changing of parameter value
3. There are actions with a lot of parameters, which will result in long names of actions.

The name of the AP could definitely be incorporated, but would it be useful? Do you want to achieve better orientation within editor or script? Because inside the editor, there could probably be found better ways to indicate conjunction with AP, based on visualization instead of action name. 

@NelaMm As you were testing the editor, could you please provide us with some examples of user friendly action names (together with name of related AP and action parameters or whatever you find useful) to help us find out what do you consider to be helpful?

ZdenekM, 2021-06-16:
Right now, the action name is equal to the action type (there might be a suffix to make the name unique). If this is not ok, could you provide some suggestions @NelaMm?

martincervinka, 2021-06-29:
The suffix should be OK.

closed 2021-06-29 by ZdenekM