robofit/arcor2_areditor #47: Enable show actual position of the end-effector

NelaMm, 2020-04-10:


Kapim, 2020-04-15:
Do you mean all the time show position of EE, or saved position/orientation withing action point, or something else?

NelaMm, 2020-04-15:
I mean all the time show the position of EE. It helps to orient in the scene. For example, sometimes I rotate the scene and then forget the starting position. The EE position helps me orient in the scene and connection between the real scene with the robot and the scene in the AR editor. It is difficult to imagine the relationship between the scene and the robot when the robot isn't show.
Maybe for each EE can be some checkbox, indicate that the EE position have to be show in scene.

Kapim, 2020-04-15:
Ok, for the "computer" version of editor this sounds reasonable, we will probably implement this once the URDF will be available through REST

Kapim, 2020-04-15:
Comment from duplicated issue #56

[image]

Kapim, 2020-04-27:
Implemented in 0.5.0

closed 2020-04-27 by Kapim