robofit/arcor2_areditor #55: Implement "Move to AP"

rstourac, 2020-04-15:
**Abstract**
It is difficult to orient in difficult scenes with many collision objects and APs. We need to introduce tools to improve this.

**Extension**
Operator wants to be able to TRY to move to selected AP to make himself sure that it is the right AP he wants to edit. It should be easy to interrupt the move at any moment. 
(For example Robots move just until user holds "Move to AP" button).

Kapim, 2020-04-15:
At the moment, the user may add action MoveTo to any action point and execute it individually within the editor. On the other hand, this will probably be very common task, so we probably can add this functionality to AP menu as well, to simplify it. 

This will probably need some changes on robot REST API, as currently (if i'm not mistaken) it is not possible to control execution of move action and stop the robot.

rstourac, 2020-04-15:
Handling the issue via special MoveTo action is workaround so I agree with special added function for it (BTW even this workaround doesn't work now see #60). 
Yes, I think that it is topic for discussion how to achieve described bahavior. Let's talk more about it on our regular meeting.

Kapim, 2020-04-15:
OK, let's talk about it tomorrow.

ZdenekM, 2020-04-16:
@Kapim There is possibility to stop robot motion (see https://gitlab.com/kinalisoft/test-it-off/robot-simulator/-/blob/master/swagger.json, tthere is PUT /robots/{robotId}/stop). Editor probably might have easily available STOP button shown whenever the robot moves?

@rstourac Movement to a pose (given by a position of the action point and a selected orientation) or to joints is definitely a good feature. Moreover, there will be visualization of stored orientations and joints (as soon as we will be able to get robot's model). Visualization will be probably in most cases sufficient and users will not need to use "move to AP" feature (checking using robot movement would be more time demanding).

rstourac, 2020-04-16:
User will always appreciate to have this feature available. It is quite common that you realize when testing the movement that it is not exactly what you expected so you want to come back to some position and define the movement from there differently. That's why movement to original position (AP), trim it a bit and update original AP's position makes sense.

Kapim, 2020-04-16:
@ZdenekM  ok, i was not aware of that. We can make stop button, but it will not be very safe (connection issues, lags, etc.). And if we would want to implement this as a dead-man-trigger, we would definitely need some changes on the API.

ZdenekM, 2020-10-21:
Already implemented.

closed 2020-10-21 by ZdenekM