robofit/arcor2_areditor #57: Make possible to visualize robot in his current position and rotation

rstourac, 2020-04-15:
**Abstract**
It is difficult to aim collision objects without proper knowledge where end-effector point is located relatively to the robot and collision object, which is about to be aimed. 

**Extension**
Operator wants to be able to see whole robot in its current position and orientation, end-effector should be highlighted as well, which allows him to get better idea where he is placing collision object exactly.
Feature should be possible switch on/off.

ZdenekM, 2020-04-16:
There is definitely a plan to be able to visualize a robot. This is certainly needed in PC application, VR mode of tablet app, even in some cases when using AR (visualization of stored joint configurations). However, robot API does not provide robot's model (an URDF) at the moment.

Kapim, 2021-02-19:
When URDF is available, the robot model is now visualized in both PC and tablet app.

closed 2021-02-19 by Kapim