robofit/arcor2_areditor #60: Execution Action button doesn't work

rstourac, 2020-04-15:
**Steps to reproduce:**

1. Create Project with at least 2 AP and Move actions to let robot move in between. 
2. Run project by hitting "Run Project" button. It works properly and robots move as expected.
3. Stop project, select one move action and try to press Execute Action.

**Observation:**
Nothing happens after pressing the button. In logs is visible that some strange values were used for Move API call. Robot usually cannot reach those positions.

Kapim, 2020-04-15:
We'll take a look on it.

ZdenekM, 2020-04-16:
This was probably caused by a bug in AR Server - during project editing, relative poses were not properly transformed into absolute before sending to the robot. The bug was fixed in https://github.com/robofit/arcor2/commit/476f2292f7ee512acd2a3e29019fdc2a2e4ba772.

closed 2020-04-16 by ZdenekM

rstourac, 2020-04-16:
I'm not sure as I didn't have relative poses in the project, just absolute...

ZdenekM, 2020-04-16:
Current version supports only relative APs - each AP belongs to some object. Absolute (global) APs will be introduced in the next version.