robofit/arcor2_areditor #61: Robot cannot reach positions when AP is placed to Project scene by mouse

rstourac, 2020-04-15:
**Steps to reproduce:**

1. Create Project with at least 2 AP and Move actions to let robot move in between. 
2. Run project by hitting "Run Project" button. It works properly and robots move as expected.
3. Stop project
4. Add one more AP and change its position by mouse a bit
5. Add new Action so robots move between not 2, but all 3 points.
6. Run project

**Observation:**
Nothing happens when robot is supposed to move to the newest AP. In logs is visible that some strange values were used for Move API call. Robot usually cannot reach this position.

Kapim, 2020-04-15:
If AP is not focused using robot, it has some default orientation (same orientation as the object for which the AP was created, as the orientation is relative), so it probably ends up with imposible configuration. The only way, at the moment, how to set orientation, is by using robot. Once the AP is focused using robot, it should be possible to move it using mouse, as this will only change its position and not orientation.

rstourac, 2020-04-15:
We didn't check orientation values so far. We saw that position values were completely wrong although it was presented correctly in Project editor.

Kapim, 2020-04-15:
@ZdenekM can you take a look on this?

ZdenekM, 2020-04-16:
@rstourac Could you please provide all relevant JSONs (scene, project) and outputs from terminal (if there are any)?

rstourac, 2020-04-24:
We were testing this issue in version 0.4.1 and the feature of positioning an AP by mouse is no longer available. System expects to choose Orientation for Move actions and that one can be created just with help of robot. So should we close this issue?

Kapim, 2020-04-24:
At the moment, we offer no way to set orientation of the AP by mouse - this will probably change in next release, because in latest builds (not released yet) we support visualization of AP orientations. The orientation thus can be set only by robot, but the position of AP can be set using mouse (using gizmo just like with action objects). So you can set orientation using robot first and then move with AP using mouse.

rstourac, 2020-04-24:
OK, but what is this for? If I set Orientation by robot it stores angles of all joints. IT means that changing position of AP has no effect. Is this just to make better overview on visual situation?

Kapim, 2020-04-24:
It stores not only joints, but also an actual orientation of the endeffector. Of course, when you move with the AP afterwards, the joints are not valid anymore, but the orientation is, so you can use it with Move action. The aiming with the robot is probably only valid option for precise manipulation at the moment, but later on, some better move feature will be introduced, which will enable for example movement of AP by defined amount of milimeters or something like that.

rstourac, 2020-04-24:
Then it doesn't work like that currently. We created new AP and setup the Move action to this AP. We chose Orientation from different AP there and robot thread those AP as the same - it didn't move when travelling between them.

Kapim, 2020-04-24:
Ok, we will take a look on that. Can you please send me logs from the editor (hamburger menu in project editor, app control, save logs - it should print the directory where logs are stored)?

rstourac, 2020-04-24:
[2020-24-4--15-09-14[3 action points].zip](https://github.com/robofit/arcor2_editor/files/4529043/2020-24-4--15-09-14.3.action.points.zip)

ZdenekM, 2020-04-27:
We found a bug in the server's code responsible for transformations between relative and absolute poses. It is already fixed and we will release a new version soon.

Kapim, 2020-04-27:
Fixed in 0.5.0

closed 2020-04-27 by Kapim