robofit/arcor2_areditor #63: MissingReferenceException after "Stop project"

ZdenekM, 2020-04-16:
MissingReferenceException occurs after a running project is stopped.

Version: https://github.com/robofit/arcor2_editor/commit/f686181cf17cf77cd217523dc5823ef2270c3cb6.

[image]

```
MissingReferenceException: The object of type 'MeshRenderer' has been destroyed but you are still trying to access it.
Your script should either check if it is null or you should not destroy the object.
UnityEngine.Renderer.get_material () (at /home/builduser/buildslave/unity/build/Runtime/Export/Graphics/GraphicsRenderers.bindings.cs:133)
Action3D.StopAction () (at Assets/TABLET/Scripts/Action3D.cs:29)
Action3D.OnProjectStop (System.Object sender, System.EventArgs e) (at Assets/TABLET/Scripts/Action3D.cs:21)
Base.GameManager+<StopProject>d__151.MoveNext () (at Assets/BASE/Scripts/GameManager.cs:706)
--- End of stack trace from previous location where exception was thrown ---
System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () (at <437ba245d8404784b9fbab9b439ac908>:0)
System.Runtime.CompilerServices.AsyncMethodBuilderCore+<>c.<ThrowAsync>b__6_0 (System.Object state) (at <437ba245d8404784b9fbab9b439ac908>:0)
UnityEngine.UnitySynchronizationContext+WorkRequest.Invoke () (at /home/builduser/buildslave/unity/build/Runtime/Export/Scripting/UnitySynchronizationContext.cs:150)
UnityEngine.UnitySynchronizationContext:ExecuteTasks() (at /home/builduser/buildslave/unity/build/Runtime/Export/Scripting/UnitySynchronizationContext.cs:104)
```

Kapim, 2020-04-17:
Can't reproduce it.. @ZdenekM is it still an issue with current version?

ZdenekM, 2020-04-17:
It still happens.

```
MissingReferenceException: The object of type 'MeshRenderer' has been destroyed but you are still trying to access it.
Your script should either check if it is null or you should not destroy the object.
UnityEngine.Renderer.get_material () (at /home/builduser/buildslave/unity/build/Runtime/Export/Graphics/GraphicsRenderers.bindings.cs:133)
Action3D.StopAction () (at Assets/TABLET/Scripts/Action3D.cs:29)
Action3D.OnProjectStop (System.Object sender, System.EventArgs e) (at Assets/TABLET/Scripts/Action3D.cs:21)
Base.GameManager+<StopProject>d__153.MoveNext () (at Assets/BASE/Scripts/GameManager.cs:722)
--- End of stack trace from previous location where exception was thrown ---
System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () (at <437ba245d8404784b9fbab9b439ac908>:0)
System.Runtime.CompilerServices.AsyncMethodBuilderCore+<>c.<ThrowAsync>b__6_0 (System.Object state) (at <437ba245d8404784b9fbab9b439ac908>:0)
UnityEngine.UnitySynchronizationContext+WorkRequest.Invoke () (at /home/builduser/buildslave/unity/build/Runtime/Export/Scripting/UnitySynchronizationContext.cs:150)
UnityEngine.UnitySynchronizationContext:ExecuteTasks() (at /home/builduser/buildslave/unity/build/Runtime/Export/Scripting/UnitySynchronizationContext.cs:104)
```

ZdenekM, 2020-04-17:
Steps to reproduce:

1. Open project.
2. Close project.
3. Open project again.
4. Run project.
5. Stop project.

Kapim, 2020-04-17:
Fixed in #85

closed 2020-04-17 by Kapim