robofit/arcor2_areditor #91: ArgumentOutOfRangeException after "Save parameters" on action with dynamic parameters

ZdenekM, 2020-04-21:
This happens when I change some parameter of e.g. RestRobotService/move and click on "Save parameters".

```
ArgumentOutOfRangeException: Index was out of range. Must be non-negative and less than the size of the collection.
Parameter name: index
System.ThrowHelper.ThrowArgumentOutOfRangeException (System.ExceptionArgument argument, System.ExceptionResource resource) (at <437ba245d8404784b9fbab9b439ac908>:0)
System.ThrowHelper.ThrowArgumentOutOfRangeException () (at <437ba245d8404784b9fbab9b439ac908>:0)
System.Collections.Generic.List`1[T].get_Item (System.Int32 index) (at <437ba245d8404784b9fbab9b439ac908>:0)
Base.Action.ActionUpdate (IO.Swagger.Model.Action action, System.Boolean updateConnections) (at Assets/BASE/Scripts/Action.cs:83)
Base.GameManager.ActionUpdated (IO.Swagger.Model.Action projectAction) (at Assets/BASE/Scripts/GameManager.cs:414)
Base.WebsocketManager.HandleActionChanged (System.String data) (at Assets/BASE/Scripts/WebsocketManager.cs:396)
Base.WebsocketManager.HandleReceivedData (System.String data) (at Assets/BASE/Scripts/WebsocketManager.cs:226)
Base.WebsocketManager+<Update>d__18.MoveNext () (at Assets/BASE/Scripts/WebsocketManager.cs:128)
--- End of stack trace from previous location where exception was thrown ---
System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () (at <437ba245d8404784b9fbab9b439ac908>:0)
System.Runtime.CompilerServices.AsyncMethodBuilderCore+<>c.<ThrowAsync>b__6_0 (System.Object state) (at <437ba245d8404784b9fbab9b439ac908>:0)
UnityEngine.UnitySynchronizationContext+WorkRequest.Invoke () (at /home/builduser/buildslave/unity/build/Runtime/Export/Scripting/UnitySynchronizationContext.cs:150)
UnityEngine.UnitySynchronizationContext:ExecuteTasks() (at /home/builduser/buildslave/unity/build/Runtime/Export/Scripting/UnitySynchronizationContext.cs:104)
```

Kapim, 2020-04-21:
Happens when there is no output connection on the action. Will be fixed for 0.4.1 release.

Kapim, 2020-04-21:
Fixed

closed 2020-04-21 by Kapim