robofit/arcor2_areditor #93: Untie action point

ZdenekM, 2020-04-21:
When I untie an action point, its context menu is not updated and shows "global action point" after it is closed and opened again. Visual connection with the parent object is not removed.

ZdenekM, 2020-04-21:
Moreover, when I untie AP, save and close project, following NullReferenceException happens:

``
NullReferenceException: Object reference not set to an instance of an object
Base.ActionPoint.DeleteAP () (at Assets/BASE/Scripts/ActionPoint.cs:188)
Base.Scene.RemoveActionPoints () (at Assets/BASE/Scripts/Scene.cs:437)
Base.GameManager+<ProjectUpdated>d__150.MoveNext () (at Assets/BASE/Scripts/GameManager.cs:624)
--- End of stack trace from previous location where exception was thrown ---
System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () (at <437ba245d8404784b9fbab9b439ac908>:0)
System.Runtime.CompilerServices.AsyncMethodBuilderCore+<>c.<ThrowAsync>b__6_0 (System.Object state) (at <437ba245d8404784b9fbab9b439ac908>:0)
UnityEngine.UnitySynchronizationContext+WorkRequest.Invoke () (at /home/builduser/buildslave/unity/build/Runtime/Export/Scripting/UnitySynchronizationContext.cs:150)
UnityEngine.UnitySynchronizationContext:ExecuteTasks() (at /home/builduser/buildslave/unity/build/Runtime/Export/Scripting/UnitySynchronizationContext.cs:104)
```

Kapim, 2020-04-21:
Fixed

closed 2020-04-21 by Kapim