robofit/arcor2_areditor #104: Unknown start action after reopening project

rstourac, 2020-04-24:
**Steps to reproduce:**

1. Create Project with at least 2 AP and Move actions to let robot move in between. 
2. Run project by hitting "Run Project" button. It works properly and robots move as expected.
3. Stop project
4. Add one more AP and change its position by mouse a bit
5. Add new Action so robots move between not 2, but all 3 points.
6. Save project
7. Close AREditor
8. Open AREditor again and reload project
9. Run project

**Observation:**
System refuses to start with message that Start action is missing.

[2020-24-4--14-48-03[unknown start action].zip](https://github.com/robofit/arcor2_editor/files/4529063/2020-24-4--14-48-03.unknown.start.action.zip)

ZdenekM, 2020-04-27:
It seems that project.json is somehow inconsistent. There are just two actions defined (move, move_2) and one of the actions (move_2) has non-existent action as an input. We were not able to reproduce this so far. Please let us know if it happens again.

rstourac, 2020-04-27:
So you were able to Run the project without any error?

ZdenekM, 2020-04-27:
The project cannot be run - it contains a reference to a non-existent action. We were not able to reproduce the situation, when such inconsistency appears in the project.

rstourac, 2020-04-27:
You cannot reproduce it even when following steps above? The point is that it works properly until you Save the project and reload it after AREditor restart. Then the problem occurs.

ZdenekM, 2020-04-27:
We (me and @Kapim) definitely followed the steps you provided. Unfortunately, without success (newly created project using the steps can be executed) - I tried it several times. One more question - you saw there three action points with one action on each of them? Because in project.json, there are just two action points.

ZdenekM, 2020-05-12:
@rstourac Is this still an issue?

rstourac, 2020-05-12:
Unfortunately we still don't have the latest version up and running so I cannot check it.

ZdenekM, 2020-06-22:
@rstourac I will close this for now. If the issue occurs again, feel free to re-open.

closed 2020-06-22 by ZdenekM