robofit/arcor2_areditor #112: Publish - catastrofic system error

NelaMm, 2020-05-07:
Maybe problem between communication between project and build
[2020-07-5--15-23-32.zip](https://github.com/robofit/arcor2_editor/files/4593204/2020-07-5--15-23-32.zip)

ZdenekM, 2020-05-08:
Certainly. Build service could not connect to the Project service. Was it running? Was ```ARCOR2_PERSISTENT_STORAGE_URL``` set? From

```
arcor2-build                    | urllib3.exceptions.MaxRetryError: HTTPConnectionPool(host='0.0.0.0', port=11000): Max retries exceeded with url: /project/32e354e3af3c4e73b6a3586771bae8b1 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fa422ecc610>: Failed to establish a new connection: [Errno 111] Connection refused'))
```
it seems that Build unit tried to communicate with Project within the same container - ```ARCOR2_PERSISTENT_STORAGE_URL``` was probably not set and the default value was used instead.

ZdenekM, 2020-05-15:
@NelaMm Is this still an issue?

NelaMm, 2020-05-15:
As you wrote, it was a bad setup. Sorry I forgot to answer.

closed 2020-05-15 by NelaMm