robofit/arcor2_areditor #114: Stop button

ZdenekM, 2020-05-08:
- AREditor should enable user to stop robot movement. However, user should be somehow warned that it is not fail-safe solution.
- Probably it could be a big red button somewhere in the corner (easily reachable), displayed only when robot moves?

ZdenekM, 2020-05-08:
@Kapim Maybe this can be generalized - it does not apply only to robots (e.g. user might want to cancel movement of arbitrary machine or even printing of a label). We can add to the action metadata information whether the action can be cancelled and there will be RPC to cancel executed action. Server will get name of the method to call in order to cancel action in a similar way to dynamic parameters (there will be dict with mapping of actions to method responsible to cancelling actions). What do you think?

Kapim, 2020-05-15:
Implemented

closed 2020-05-15 by Kapim