robofit/arcor2_areditor #134: Build and run - position of EE

NelaMm, 2020-06-05:
After open project and click Build and run is project closed. Is it expected?
And project is builded and runned, after click stop, it act undefined. Sometimes is opened project but position of EE is on bad position(on center), sometimes is still on script view, but can't click close (only stop/pause script, bud it was stopped). Editor must be closed and opened. 

May be it is caused by calling of creating or destroying system and not waiting for response or calling in quick succession (these actions are extremelly time consuming and calling in sequence may caused problems or undefined act).

These action couldn't be call after Build and run because if is project open these action was called on open and after stop script we can't destroy system when project is still opened and system should have be still created. Or call destroy on stop but call create on opened project. You can call destroy only on close project or close scene.

Kapim, 2020-06-11:
> After open project and click Build and run is project closed. Is it expected?

Build and run literally means: Build project, close project, run project. As the initialization of robot takes 10s, the system looks strange - we will add loading screen to inform the user that "something" is happening.

> And project is builded and runned, after click stop, it act undefined. Sometimes is opened project but position of EE is on bad position(on center), sometimes is still on script view, but can't [...]

Ok, we will check this

ZdenekM, 2021-06-16:
I guess that this issue is outdated and can be closed.

closed 2021-06-16 by ZdenekM