robofit/arcor2_areditor #138: Object name visualization

ZdenekM, 2020-06-17:
Object name label should be independent on object's dimensions and rotation. It should (probably) behave like a billboard - be always oriented towards user?

[image]

xBambusekD, 2020-07-08:
Updated in #161:
[image]

But issues with text overlapping occured (_tester_ name not hidden by box object):
[image]

ZdenekM, 2020-07-09:
@xBambusekD It looks good!

closed 2020-07-29 by Kapim