robofit/arcor2_areditor #150: Marker calibration: multiple cubes, offset

ZdenekM, 2020-07-06:
AREditor version: 0.7.2.

- After calibration (clicking on the cube), there is always some offset - tried on different surfaces.
- Sometimes, there are more cubes.

[image]

[image]

[image]

[image]

xBambusekD, 2020-07-08:
For best calibration results, you should click on the calibration cube with the camera pointing straight down on the marker, otherwise there will always be some offset due to imperfect image processing of ARCore.
[image]

Regarding multiple cubes, can you describe steps on how to reproduce this issue?

ZdenekM, 2020-07-09:
Hmm, today I had troubles with reproducing this. But I found one way, see video: https://photos.app.goo.gl/CSvbie9zzEiVgrYF6. 

[logs.txt](https://github.com/robofit/arcor2_editor/files/4894982/logs.txt)

closed 2021-03-04 by ZdenekM