robofit/arcor2_areditor #162: Some texts in status line / menus are hidden by virtual objects

ZdenekM, 2020-07-09:
AREditor version: 0.7.2

See video: https://photos.app.goo.gl/xF5d2HEoR1SBbCHw7

Kapim, 2020-07-13:
@xBambusekD you fixed this? I can't reproduce this in your latest devel build..

xBambusekD, 2020-07-13:
I wasn't working on this bug at all. It was probably fixed unpurposely by changing the shader of LiberationSans SDF Material to TextMeshPro/Mobile/Distance Field Overlay (for object names display). Shader switch on this font changed it on all texts using this font, and as I am looking, this font is used almost everywhere. So I assume, it is fixed.
[image]

Kapim, 2020-07-13:
Nice :)

closed 2020-07-13 by Kapim