robofit/arcor2_areditor #181: Show/hide calibration cube does not work

NelaMm, 2020-11-09:
Show/hide calibration cube does not work on tablet

**Found in version**
Editor version: 0.10.2
Server version: 0.10.1
Editor API version: 0.9.2
Server API version: 0.9.2

**Perform the following steps reproduce the bug**
- Install application to tablet
- Connect to arserver
- Create new scene
- Click Show/hide calibration cube

**Observed results**
Exception occured

**Expected results**
Show calibration cube

[logs.txt](https://github.com/robofit/arcor2/files/5501264/logs.txt)

ZdenekM, 2020-11-09:
@NelaMm I transferred this issue to arcor2_editor repo.

ZdenekM, 2020-11-09:
I can confirm the issue. It happens only with a new scene. When I open the existing scene, the button/cube works as expected. @xBambusekD, could you take a look, please?

NelaMm, 2020-11-09:
> I can confirm the issue. It happens only with a new scene. When I open the existing scene, the button/cube works as expected. 

When I opened an existing scene after caused this exception, it caused again. Also after uninstalling and reinstalling the application. No scene could be calibrated at this time.

ZdenekM, 2020-11-09:
Btw, with a newly created scene, it is not necessary to click on the calibration cube icon - it is switched on (at least in my case) by default. The cube appears by itself when tablet sees the calibration marker. Then, I'm able to perform calibration. After calibration, the cube can be shown/hidden without any troubles.

ZdenekM, 2020-11-09:
So the workflow should be:

- Install application to tablet
- Connect to arserver
- Create new scene
- **Locate calibration marker, wait until cube appears and click on it**
- Click Show/hide calibration cube

Are you doing it like this?

NelaMm, 2020-11-09:
Thanks for detailed instruction. Sorry, it was my mistake. The problem was caused by old and scratched calibration markings, so the cube does not appear.

xBambusekD, 2020-11-09:
@NelaMm Thank you for this issue. There was a null reference error, which should be fixed now. I also disabled the show/hide calibration cube button, when the system is not calibrated yet. Upon clicking that button, a notification will pop up saying that you must calibrate first (find marker, wait for the cube to appear and click on it). When the system is calibrated, the show/hide button works as expected.

closed 2020-11-09 by xBambusekD