robofit/arcor2_areditor #184: The connections between the actions is not displayed on tablet

NelaMm, 2020-11-10:
The connections between the actions is not displayed. Cannot manipulate with connetions from tablet (OS Android 10).

**Found in version**
Editor version: 0.10.3
Server version: 0.10.1
Editor API version: 0.9.2
Server API version: 0.9.2

**Perform the following steps reproduce the bug**
- Install application to tablet
- Connect to arserver
- Open project with actions and connections

**Observed results**
The connections between the actions is not displayed.

**Expected results**
The connections between the actions is displayed.

Related data

[image]
[image]

Kapim, 2020-11-10:
There is a button in the control box (bottom side of the screen), which enables and disables visibility of connections. 

[image]

I'm sorry, I did not realize it could be the issue when I wrote you the email before.

NelaMm, 2020-11-10:
Yes, i noticed that, but it still doesn't work anyway as in version 0.10.2.

Kapim, 2020-11-10:
No, dark gray means it is pressed, i.e. enabled. It is probably not very intuitive. We will take this into account in the upcoming redesign. In the first image you send in the first post, it is enabled, thus connections are visible. In the second one, it is not pressed and connections are invisible.. I just checked here with 0.10.3 and it is working as expected, could you please send me logs if it is still invisible with this button pressed?

NelaMm, 2020-11-10:
I tried both mode yesterday and today and it was still invisible, but now it show up. I don't get it. Now I look like crazy. But I hope something is in the logs. 
[Logs.zip](https://github.com/robofit/arcor2_areditor/files/5517996/Logs.zip)

[2020-09-11--15-14-32.zip](https://github.com/robofit/arcor2_areditor/files/5517882/2020-09-11--15-14-32.zip)

Kapim, 2020-11-10:
Don't worry, you are not crazy :) based on the logs I just found that there really is an issue with the "Display connections" button. If you hide connections, then close and reopen project, the button will stop working. If you set it back to show connections and close and reopen project, it will work again..  Thanks for pointing this out, we will fix it for next release.

[image]

Kapim, 2020-11-18:
Fixed for the next release.

closed 2020-11-18 by Kapim