robofit/arcor2_areditor #186: After change parameter 'save' is not able to save the parameter

NelaMm, 2020-11-12:
After change parameter _save_ from enable to disable in method _set_joints_ does not enable to save the parameter. Button  _Save parameters_ is disabled and show message Parameters unchanged.

**Found in version**
Editor version: 0.10.3
Server version: 0.10.1
Editor API version: 0.9.2
Server API version: 0.9.2

**Perform the following steps reproduce the bug**
- Install application to tablet
- Connect to arserver
- Create scene
- Calibrate scene
- Add robot service
- Save scene and create project
- Add global AP
- Start scene
- Focus AP by robot
- Add method set_joints, fill parameters and enable parameter save
- Save project
- Edit method set_joints  and change parametr save

**Observed results**
- Button  Save parameters is disabled and after click show message Parameters unchanged.

**Expected results**
- Button Save parameter enable to save parameters.

Kapim, 2020-11-18:
Thanks for reporting, I will take a look on that.

Kapim, 2020-11-18:
Fixed, will be added to next release.

closed 2020-11-18 by Kapim