robofit/arcor2_areditor #197: Make entering names easier

ZdenekM, 2021-02-12:
As there are some restrictions for ObjectType type (which is actually class name in Python) and AP/object/etc. names (which are actually variables/properties), AREditor should provide more guidance when entering those:
- Show example.
- Disable disallowed characters on the keyboard.
- Something else?

This is related to https://github.com/robofit/arcor2/issues/443.

ZdenekM, 2021-02-12:
It seems that Unity/Android does not have support for limiting available characters on the keyboard. Maybe custom validation for the input field might help here? See https://docs.unity3d.com/540/Documentation/ScriptReference/UI.InputField.OnValidateInput.html.

ZdenekM, 2021-02-16:
@Kapim ARServer behavior changed (the issue was closed). But I just got an idea, that it would be good if ARServer can provide a more specific error message on what is wrong with the type/name, maybe?

Kapim, 2021-02-16:
I depends.. If there will be custom validator in editor, detailed error message should not be needed, because no wrong input will be passed to server (except for already used name). On the other hand, if there will be detailed message, no custom validation will be needed in AREditor or any other (future) GUI.

ZdenekM, 2021-02-16:
I think that the validator in the editor could be very very simple - probably just banning infeasible characters (if it is possible). The rest (e.g. checking if it is not a Python keyword) should be done by the server and the server should provide better error messages. Ok?

ZdenekM, 2021-02-16:
Better error messages: https://github.com/robofit/arcor2/pull/450.